toy/blueutil #84: Incorrect Recent Access Date

dalegallosky, 2024-01-26:
Hello, 

When attempting --paired I see all my devices but the recent access date all shows the current date/time. Using --info with the MAC address results in similar results. Terminal access is toggled on for both Developer Tools and Bluetooth, and I restarted Terminal. 

MacOS 14.3.

Thank you!

[image]

toy, 2024-01-27:
The value for recent access date is coming directly from framework, code just gets paired devices and prints the values. I can suggest trying to restart and check if possible on different machine.

dalegallosky, 2024-01-31:
Thanks - confirmed issue occurs on three different machines - two Apple silicon; one Intel.

toy, 2024-01-31:
I've checked on available machines and seems it got broken in the framework between Catalina and Monterey. Looks like only submitting a bug to Apple can help with this.

toy, 2024-04-10:
While checking for #89 I thought that it is probably not a bug, but either privacy feature or just functionality that was silently removed, so I'll document it

commit 30fbc488 (2024-04-10): document macOS >= 12 silently ignoring favourites and recent access date

Resolves #63
Resolves #84
closed 2024-04-10 by toy