toy/blueutil #85: Certain flags do not work over ssh

jwhite007, 2024-03-12:
blueutil v2.9.1 installed via Homebrew on Sonoma 14.4

Firstly, I'd like to thank the developer(s) for this wonderful tool.

I've used blueutil in the past over ssh without issue. In fact, that is my main use case for blueutil. If I want to connect to a bluetooth device from a local mac but another mac in another room is connected, it's much easier just to ssh into that other mac and disconnect the bluetooth device so that I can connect on the local mac. For some reason, this latest version is not fully functional over ssh, though. 

Of the flags that I've tried, the following do not work.
```
$ blueutil --paired  # returns nothing even when local it returns values
$ blueutil --connected  # returns nothing even when local it returns values
$ blueutil --disconnect <ID>   # hangs
```

the following works
```
$ blueutil -p
1
```

toy, 2024-03-12:
Thank you and glad it was (and hopefully will be) useful!

I suspect that os privacy restrictions are causing this.
They are inherited from the ancestor which is the child of launchd, so for terminal you must have permission to use bluetooth, but not for sshd. If this is the case, you need to add sshd to the list and give it permission, then restart sshd and you should be able to have same access as from terminal.

jwhite007, 2024-03-13:
@toy ,

Thanks for the reply; however, it doesn't work for me. 

[image]

followed by:
```
$ sudo launchctl unload  /System/Library/LaunchDaemons/ssh.plist
$ sudo launchctl load -w /System/Library/LaunchDaemons/ssh.plist
```

toy, 2024-03-13:
After some checking it maybe not sshd itself, but `sshd-keygen-wrapper` (`/usr/libexec/sshd-keygen-wrapper`) that you need to add and grant access to

jwhite007, 2024-03-14:
@toy,

That did the trick. Thank you, sir, for all that you've done with blueutil.

jwhite007, 2024-03-14:
@toy ,

That did the trick. Thank you, sir, for all that you've done with blueutil.

closed 2024-03-14 by jwhite007

milindprajapati, 2024-04-11:
@toy 

Thanks this helps. Although when I run inquiry, it get stuck and shows
 2024-04-11 11:21:33.699 blueutil[6056:425757] -[IOBluetoothDeviceInquiry initWithDelegate:] -  0x600003244030
Would you be able to help me out?

toy, 2024-04-11:
@milindprajapati I've answered in #60

stuartbrussell, 2025-04-22:
@toy 
blueutil was not working over ssh on my Sequoia machine, adding `sshd-keygen-wrapper` to Bluetooth permissions worked. Thank you!