toy/blueutil #87: Bose QC Ultra won't show up as connected

m1keil, 2024-03-22:
This is just another data point similar to #82.

The device appears in the `--paired` list but will show up as "not connected" and won't appear in the `--connected` list even when it is actually connected and visible in the regular MacOS bluetooth devices list. 

Interestingly enough, doing `--connect <id>` does work and the device connects to the laptop but still appears as "not connected" via blueutil.

This seem to be replicated across two different Macbooks (M1 Pro and M2 Pro).

I wonder if this is somehow related to the fact that these headphones support multi-device pairing?

toy, 2024-03-22:
Did you give bluetooth permission to your terminal? (I guess yes, given that you are able to connect a device)

But I doubt that it is related to the headphones ability to pair with multiple devices.

m1keil, 2024-03-23:
Yes, terminal has bluetooth permission. Tried in both iTerm and the stock terminal.app (which prompted me to allow access to bluetooth before first use).

Let me know if you want me to attempt any further debug steps.

toy, 2024-03-23:
Unfortunately no good ideas what to debug currently.
I checked the headphones I have that can pair with two devices and they are displayed connected by blueutil both when audio comes from the mac and from the other device.
If you have access to some other bluetooth headphones, especially ones that can pair/connect to multiple devices and you can check if they will be displayed connected. Yes would be suggestion towards an issue with headphones, no towards os/framework, but still a very vague/circumstantial one.

megalithic, 2024-12-31:
@m1keil I'm experiencing this as well. I've not been able to find a solution.

JoachimThomas, 2025-02-17:
Same here with Bose Soundbar 300. 

[image]

arcaartem, 2025-07-23:
I have the same headphones as @m1keil  and I'm observing the same behaviour as described. I'm happy to spend some time to debug the issue. @toy do you have any intuitions which area the problem potentially might be in?

toy, 2025-07-25:
> [@toy](https://github.com/toy) do you have any intuitions which area the problem potentially might be in?

@arcaartem My understanding is that IOBluetooth is just not fully maintained to properly expose all bluetooth devices in os and it is like that for multiple os versions
Thank you for the PR, I'll try to look at it soon