toy/blueutil #89: Blueutil paired, recent did not work - Until Checking Discoverable

NelsonScott, 2024-04-07:
Steps to Reproduce:
1) Install blueutil w/ homebrew on mac Sonoma
2) Try to see paired, recent. Get no results.
3) Check discoverable
4) Now paired seems to work 

```
~ » blueutil --paired                                                                                                                                                                                                               08:24:04
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
~ » blueutil recent                                                                                                                                                                                                                 08:24:21
Unexpected arguments: recent
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
~ » blueutil --recent                                                                                                                                                                                                               08:24:28
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
~ » blueutil --recent                                                                                                                                                                                                               08:24:33
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
~ » blueutil --discoverable                                                                                                                                                                                                         08:24:47
0
```

toy, 2024-04-07:
As far as I know osx is not remembering recent devices since some versions ago, but listing paired should work as long as terminal has bluetooth permission. Also discoverable should work even without bluetooth permission.

Can you repeatedly reproduce it (after reinstalling blueutil or restart) or did it just not work one after installing blueutil?

NelsonScott, 2024-04-09:
So i only did it once, this was a fresh install.  Just provided what I did
in case someone wants to reproduce.

Just checking again and yeah recent seems to never return anything (should
it? if not might want to make that clear), but 'paired' does work.

toy, 2024-04-10:
Good point about mentioning that recent is not recorded by os

Concerning paired - probably better to let the issue stay open for few weeks if someone has it too