toy/blueutil #90: blueuti --unpair

FeliksML, 2024-04-09:
blueuti --unpair not working on MacOs Sanoma. It just executes but don't change anything. 

LOG:

Last login: Mon Apr  8 16:18:23 on ttys000
(base) ➜  ~ blueutil --paired
address: 14-c8-8b-e7-53-74, connected (master, 0 dBm), not favourite, paired, name: "Feliks’s AirPods Max", recent access date: 2024-04-08 23:58:53 +0000
address: 88-c9-e8-35-8b-29, not connected, not favourite, paired, name: "WH-1000XM4", recent access date: 2024-04-08 23:58:53 +0000
address: a4-16-c0-95-09-7e, not connected, not favourite, paired, name: "Feliks’s AirPods Pro", recent access date: 2024-04-08 23:58:53 +0000
address: dc-2c-26-15-b8-6e, connected (master, 0 dBm), not favourite, paired, name: "Keychron K3", recent access date: 2024-04-08 23:58:53 +0000
address: d0-f9-09-96-04-cf, not connected, not favourite, paired, name: "Pebble i345", recent access date: 2024-04-08 23:58:53 +0000
address: 7c-61-30-9e-74-98, not connected, not favourite, paired, name: "Feliks’s Apple Watch", recent access date: 2024-04-08 23:58:53 +0000
address: d0-f9-09-96-04-cd, not connected, not favourite, paired, name: "Pebble i345", recent access date: 2024-04-08 23:58:53 +0000
address: d0-f9-09-96-04-d1, not connected, not favourite, paired, name: "Pebble i345", recent access date: 2024-04-08 23:58:53 +0000
address: d0-f9-09-96-04-d3, not connected, not favourite, paired, name: "Pebble i345", recent access date: 2024-04-08 23:58:53 +0000
address: d0-f9-09-96-04-d5, not connected, not favourite, paired, name: "Pebble i345", recent access date: 2024-04-08 23:58:53 +0000
address: xxxxxxxxxxxxx, not connected, not favourite, paired, name: "Feliks’s iPhone", recent access date: 2024-04-08 23:58:53 +0000
address: d0-f9-09-96-04-d7, connected (master, 0 dBm), not favourite, paired, name: "Pebble i345", recent access date: 2024-04-08 23:58:53 +0000
(base) ➜  ~ blueutil --unpair d0-f9-09-96-04-d5
(base) ➜  ~ blueutil --paired                  
address: 14-c8-8b-e7-53-74, connected (master, 0 dBm), not favourite, paired, name: "Feliks’s AirPods Max", recent access date: 2024-04-08 23:59:26 +0000
address: 88-c9-e8-35-8b-29, not connected, not favourite, paired, name: "WH-1000XM4", recent access date: 2024-04-08 23:59:26 +0000
address: a4-16-c0-95-09-7e, not connected, not favourite, paired, name: "Feliks’s AirPods Pro", recent access date: 2024-04-08 23:59:26 +0000
address: dc-2c-26-15-b8-6e, connected (master, 0 dBm), not favourite, paired, name: "Keychron K3", recent access date: 2024-04-08 23:59:26 +0000
address: d0-f9-09-96-04-cf, not connected, not favourite, paired, name: "Pebble i345", recent access date: 2024-04-08 23:59:26 +0000
address: 7c-61-30-9e-74-98, not connected, not favourite, paired, name: "Feliks’s Apple Watch", recent access date: 2024-04-08 23:59:26 +0000
address: d0-f9-09-96-04-cd, not connected, not favourite, paired, name: "Pebble i345", recent access date: 2024-04-08 23:59:26 +0000
address: d0-f9-09-96-04-d1, not connected, not favourite, paired, name: "Pebble i345", recent access date: 2024-04-08 23:59:26 +0000
address: d0-f9-09-96-04-d3, not connected, not favourite, paired, name: "Pebble i345", recent access date: 2024-04-08 23:59:26 +0000
address: d0-f9-09-96-04-d5, not connected, not favourite, paired, name: "Pebble i345", recent access date: 2024-04-08 23:59:26 +0000
address: xxxxxxxxxxxxxxxxxx, not connected, not favourite, paired, name: "Feliks’s iPhone", recent access date: 2024-04-08 23:59:26 +0000
address: d0-f9-09-96-04-d7, connected (master, 0 dBm), not favourite, paired, name: "Pebble i345", recent access date: 2024-04-08 23:59:26 +0000
(base) ➜  ~

toy, 2024-04-09:
Does it not work only with that specific device? Are you able to disconnect other Pebble i345s (or is it the same device with different addresses) or other devices?

FeliksML, 2024-04-09:
Its the same device with different addresses.

FeliksML, 2024-04-09:
I sometimes need to disconnect this device and when I want to reconnect it appears as new device with different MAC address. I want to write script which will delete the previous connection from the bluetooth list.

toy, 2024-04-10:
Please check if other devices also can't be unpaired, for example Keychron K3

xilopaint, 2024-04-27:
@toy I have the same problem. There's an old MacBook Air listed that I no longer don't own. I try to run `blueutil --unpair` to remove the device from the list, but it does not work. I also can't remove the device from System Settings because it doesn't appear there. Is there any other way of removing this device from the `blueutil` paired list?

toy, 2024-04-29:
@xilopaint Original issue is about not being able to unpair a device, but you have a device that is not listed by System Settings, only by `blueutil --paired` which is already an issue itself.

A bit of digging suggests that there are two files for paired devices:
* `/Library/Bluetooth/Library/Preferences/com.apple.MobileBluetooth.devices.plist` (apparently for classic devices)
* `/Library/Bluetooth/com.apple.MobileBluetooth.ledevices.paired.db` (by name for low energy devices)

Looks like bluetooth framework does not always delete entries from the second file and there is some inconsistency in what listed by System Settings and returned by framework (and listed by blueutil).

I'm not yet sure what could be the right way to deal with this.

xilopaint, 2024-05-02:
@toy Please let me know if I can help you to debug the issue.

shah-ensa-scacap, 2024-09-02:
Hi @toy 

I have two macs and used to switch my keyboard and touchpad between them using `blueutil` for more than a year. But it stopped working a couple days ago on the newer mac with `Sonoma 14.6.1 (23G93)`. (but still working on mac with older os and intel cpu)
In my case, both pair and upair stopped working. Do we have any update on this issue or any workaround?

toy, 2024-09-03:
@shah-ensa-scacap Unfortunately no updates for now

kassi, 2024-11-26:
Same problem here as @shah-ensa-scacap, however it doesn't even work when using the UI via system preferences.
I basically always need to forget/ignore the device, then I can pair again later after I paired with a different mac.
An option to perform this from command line would be great.

Both Macs are ARM and latest OS, but it was the same on previous OS as well.

toy, 2024-11-26:
@kassi So preference pane from apple isn't working well with a framework from apple, doesn't look good, but I still hope there will be an idea/insight in how to fix this for newer OS versions