toy/blueutil #91: Frequent switching failures between two Macs

falconws, 2024-04-17:
I am switching between a MacBook Pro 2017 (Ventura 13.6.6) and an M3 MacBook Pro (Sonoma 14.4.1) with a genuine Apple Magic Keyboard and Magic Trackpad.
The script used is as follows.

pair.sh
```zsh
#!/bin/zsh

TRACKPAD_ID=xx-xx-xx-xx-xx-xx
KEYBOARD_ID=zz-zz-zz-zz-zz-zz

blueutil --pair $TRACKPAD_ID
sleep 2
blueutil --pair $KEYBOARD_ID
sleep 2

blueutil --connect $TRACKPAD_ID
sleep 2
blueutil --connect $KEYBOARD_ID
sleep 2
```

unpair.sh
```zsh
#!/bin/zsh

TRACKPAD_ID=34-b1-eb-ec-9c-86
KEYBOARD_ID=80-95-3a-b7-46-9d

blueutil --disconnect $TRACKPAD_ID
sleep 2
blueutil --disconnect $KEYBOARD_ID
sleep 2

blueutil --unpair $TRACKPAD_ID
sleep 2
blueutil --unpair $KEYBOARD_ID
sleep 2
```
(Do I need sleep command?)

I run unpair.sh at the source and pair.sh at the destination.
When pair.sh is executed, 
```
Failed to pair "<address>" with error 0x02 (No Connection)
```
and
```
"Failed to connect "<address>"
```
are outputted and it does not switch over properly. Am I using something wrong? Any advice would be appreciated.

toy, 2024-05-01:
I thought that triggering pairing is required on device before starting pairing, but after trying it on a keyboard I have, seems that unpairing of Apple device also tells it that it is unpaired, so it automatically switches to pairing mode.
It will still probably be frustrating if you need to switch often, as you need to type a random pin on keyboard with every pairing.

About scripts - sleeps should not be needed, also I think connect will happen automatically after pairing, so no need to do it explicitly and no need to disconnect explicitly before unpairing.

falconws, 2024-05-02:
@toy 
Thank you for your advice.
I recognized the following.

1. No need sleep
2. No need `--connect` and `--disconnect`. Just need `--pair` and `--unpair` only

Is that correct?

toy, 2024-05-02:
Yes, you just need variables for device ids/addresses and pair/unpair commands.

I can also suggest that maybe generating a 6 digit number for pin (instead of coming up with one or using a fixed one) will make it easier.

falconws, 2024-05-03:
@toy 
Thanks.
But Apple Magic Keyboard and Apple Magic Trackpad aren't need pin for pairing 😄 
Thanks to toy. Solved my problem!!

closed 2024-05-03 by falconws

toy, 2024-05-03:
Hm, Magic Trackpad doesn't have an option to display or input a pin, but all keyboards should require a pin for pairing and my old Apple keyboard does require one.

I'm glad that I was able to help

falconws, 2024-05-04:
> Magic Trackpad doesn't have an option to display or input a pin, but all keyboards should require a pin for pairing

Hm.. 🤔 
Why Apple Magic Trackpad can pairing without pin?

toy, 2024-05-04:
> Why Apple Magic Trackpad can pairing without pin?

It has no display or keypad, so it can neither display nor receive numeric input.