toy/blueutil #92: General failure on Sonoma OS

dmnelson, 2024-07-01:
I'm getting an error on most commands with blueutil on a fresh install on a 2019 Intel Mac OS:

```
> blueutil --version
2.10.0

> blueutil
[1]    46278 abort      blueutil
```
It was installed using Homebrew.

As for some others such as `--recent` and `--favorites`, it doesn't error but doesn't produce any output either.

I'm happy to provide any additional information if anyone could give me any pointers of what would be most helpful.

toy, 2024-07-01:
Odd as both `blueutil` and `blueutil --recent` use the framework, so it fails only when calling certain parts of framework, maybe parts contacting hardware. There are no `abort` calls in the code, so it must be coming from framework, standard library or os.

Is there something special about your system? Does Bluetooth settings/preferences pane work for you?

Maybe check if something interesting comes up in output of `log stream --predicate 'eventMessage contains "blueutil"' --info` when calling `blueutil`.

Output of `--recent` is unfortunately expected to be empty (see [changelog](https://github.com/toy/blueutil/blob/main/CHANGELOG.md#v2100-2024-04-28)).

tgullii, 2024-07-04:
@dmnelson I ran into the same issue. Turns out all I had to do was accept a popup asking if Terminal should be granted access to Bluetooth. As I was using several monitors, this popup was hard to notice.

dmnelson, 2024-07-04:
@tgullii You are spot on! Thanks! It turns out that the popup wasn't really showing for the terminal app I use (Warp). Tried on iterm2 and it worked. Then I manually added the permission to Warp and now it's working as well.

Also thanks @toy for the assistance! With the log I was able to see the error message was related to the terminal app which was the first clue.

In the end I think the issue was not related to anything specific to blueutil, so I will close this. Thanks you both again.

closed 2024-07-04 by dmnelson