toy/blueutil #95: `abort` due to missing permissions

Tener, 2025-01-14:
When parent terminal fails to obtain Bluetooth permission, `blueutil` crashes when trying to interact with respective OS APIs. It should handle this issue gracefully instead, informing the user about it.

```
> blueutil  --connect f8-ab-e5-a7-6a-e8
[1]    38143 abort      blueutil --connect f8-ab-e5-a7-6a-e8
```

In my particular case this was the result of switching to Ghostty. I'm not sure if it failed to request this permission or I dismissed the prompt, but nevertheless `blueutil` shouldn't just crash.

toy, 2025-01-14:
Yes, informing the user is a good idea, should be easy to do

oschrenk, 2025-01-18:
Came here for the same issue. Was confused about `SIGABRT` error code when doing `blueutil --connected`.  As OP I switched to Ghostty.

To give Ghostty permission to Bluetooth, open Settings > Privacy & Security > Bluetooth > Press the little grey + button > Find Ghosty.app (should be in `/Applications`) > Restart Ghostty.

toy, 2025-01-18:
Thanks for the nudge, I just released v2.11.0