toy/blueutil #96: A quick thank you

Zaloog, 2025-01-19:
Hi @toy, 
thank you for blueutil.
I just switched to Mac and wanted something to control my bluetooth from the terminal, and blueutil worked great.

I build a minimal TUI yesterday which utilizes blueutil in the background (https://github.com/Zaloog/blueutil-tui). 
 Cause I didn't want to write the commands.
Works like a charm. Although I can't for example disconnect my keyboard, but it works great with my AirPods, which was the main motivation to build it anyway. 

[image]

So thanks again and keep it up

toy, 2025-01-25:
Thank you!
Inability to disconnect keyboard is probably #58
What do you think about adding to readme a link to the TUI you created?

Zaloog, 2025-01-25:
Thanks for the solution,
will try the solution proposed in #58 and maybe add that as a fallback into the TUI. 

Do you have any preferences where to put the reference to the TUI in your current Readme? I think after installation in the Notes section might fit or do you think a separate header is needed?
Is the  same screenshot as from the repo and a short sentence with a link to blueutil-tui ok?

toy, 2025-01-25:
I looked at the `Notes` section, wondered why is it at the bottom, so moved it higher :)
A separate section after `Installation` would be better, not sure though about the name to make it general enough, "Higher level interfaces", "Tools", ideas?
Some description and link, but probably better without screenshot, in this repo it will be odd and will become outdated, and linked will be a dependency.

Zaloog, 2025-01-30:
Named the section `Alternative Interface` and listed the current supported functionalities.

commit dae32016 (2025-01-31): Add blueutil-tui reference to README #97

closes #96
closed 2025-01-31 by toy