toy/blueutil #100: It doesn’t work on M2 Pro macOS 15.3.1 (24D70).

txunda, 2025-05-21:
It’s driving me crazy that it doesn’t work on M2 Pro macOS 15.3.1 (24D70).

jasper@Jasper-MacBook-Pro-M2 ~ % blueutil --version
2.12.0
jasper@Jasper-MacBook-Pro-M2 ~ % blueutil --connected
address: d7-70-50-02-35-b2, connected (master, 0 dBm), not favourite, paired, name: "HUAWEI Mouse CD23", recent access date: 2025-05-21 08:47:29 +0000
address: d7-a2-2b-93-14-4a, connected (master, 0 dBm), not favourite, paired, name: "HUAWEI Ultrathin Keyboard", recent access date: 2025-05-21 08:47:29 +0000
jasper@Jasper-MacBook-Pro-M2 ~ % blueutil --info d7-70-50-02-35-b2
address: d7-70-50-02-35-b2, not connected, not favourite, not paired, name: "", recent access date: 2025-05-21 08:47:40 +0000
jasper@Jasper-MacBook-Pro-M2 ~ % blueutil --is-connected d7-70-50-02-35-b2
0
jasper@Jasper-MacBook-Pro-M2 ~ % blueutil --connect d7-70-50-02-35-b2     
^C
jasper@Jasper-MacBook-Pro-M2 ~ % blueutil --disconnect d7-70-50-02-35-b2
jasper@Jasper-MacBook-Pro-M2 ~ % blueutil --disconnect d7-70-50-02-35-b2
jasper@Jasper-MacBook-Pro-M2 ~ % blueutil --disconnect d7-70-50-02-35-b2 --info d7-70-50-02-35-b2
address: d7-70-50-02-35-b2, connected (master, 0 dBm), not favourite, paired, name: "HUAWEI Mouse CD23", recent access date: 2025-05-21 08:49:45 +0000
jasper@Jasper-MacBook-Pro-M2 ~ % blueutil --disconnect 'HUAWEI Mouse CD23'
jasper@Jasper-MacBook-Pro-M2 ~ % blueutil --wait-disconnect d7-70-50-02-35-b2                    
jasper@Jasper-MacBook-Pro-M2 ~ % blueutil --disconnect d7-70-50-02-35-b2                         
jasper@Jasper-MacBook-Pro-M2 ~ % blueutil --disconnect D7:70:50:02:35:B2

toy, 2025-06-08:
`d7-70-50-02-35-b2` is reported connected by `blueutil --connected`, but not connected by `blueutil --info d7-70-50-02-35-b2` and `blueutil --info d7-70-50-02-35-b2` right after. Does it stay connected all that time?

toy, 2025-06-21:
@xa4hf8 Does disable bluetooth through menu or system preferences work?

toy, 2025-06-22:
@xa4hf8 If you can't turn off bluetooth through System Preferences, then the problem is on OS level. And just to eliminate confusion - are you certain that the mouse is bluetooth one and not other wireless using a dongle device?

toy, 2025-06-23:
@xa4hf8 You mean file transfer doesn't work when enabling bluetooth using `blueutil`, but works when using System Preferences?