toy/blueutil #103: `--disconnect` hangs when using an unknown address

darkhz, 2025-09-30:
Whenever the command `blueutil --disconnect` is used with an address that is of a valid format but is unknown by the system, the command hangs indefinitely, when tested on Sequoia. Consequently, it works correctly and returns almost immediately when a known bluetooth address is entered.

Maybe it is due to the fact that `-[IOBluetoothDevice deviceWithAddressString:]` returns a non-null IOBluetoothDevice object with the unknown address, and either `closeConnection` blocks indefinitely or the disconnect command waits for a disconnection notification that is never posted. Perhaps, the null checks for the IOBluetoothDevice in the `get_device` function isn't enough to determine whether an IOBluetoothDevice object is valid.

toy, 2025-10-04:
Thank you for report, I can confirm that for example `blueutil --disconnect 00-11-22-33-44-55` hangs (stopped waiting after 2 minutes).

Documentation of `[IOBluetoothDevice deviceWithAddressString:]` doesn't mention returning anything but device and the method is also used for pairing devices. So I assume that check for null in `get_device` was a precaution and device is always returned. Maybe the solution is to check if device is connected before trying to disconnect it

darkhz, 2025-10-04:
Hello @toy 

Thanks for reviewing this issue. 

If I may elaborate, I believe the `deviceWithAddressString:` function is buggy or maybe is working as advertised. The behaviour of this particular function stems from the `CBClassicManager` class, wherein (correct me if I'm wrong):

(Assuming that an unknown bluetooth address is provided)

a. -[IOBluetoothDevice deviceWithAddressString:] calls into [CBClassicManager retrievePeerWithAddress:].

b. [CBClassicManager retrievePeerWithAddress:] essentially tries to obtain any valid device data and initially populates an object of type `CBClassicPeer`. Important to note that this value is never null and is populated with some initial values, although if it is an unknown address, the `addressString` property of `CBClassicPeer` is empty.

c. Within the same function, whether or not the address is valid or invalid, the provided bluetooth address is forced to be set within the CBClassicPeer object. This object is then converted to an IOBluetoothDevice object. Do note that if the address was null in the CBClassicPeer object, then the IOBluetoothDevice object would be NULL. The issue is the force-setting of the bluetooth address despite it being unknown by the system.

toy, 2025-10-05:
I don't know if you are right about the reason coming from `CBClassicManager`/`CBClassicPeer`, but I definitely agree about frustrating state of documentation.
I assume https://github.com/toy/blueutil/pull/new/fix-halt-on-unknown-disconnect should fix the issue

darkhz, 2025-10-05:
The proposed fix in the pull request should work, since if an invalid IOBluetoothDevice object is returned, the `isConnected` property should be false anyway.

I can test this later and let you know. However, and this is just a suggestion, maybe evaluate other code-paths where get_device() is called and potentially introduce additional checks if needed?

Thanks for fixing this issue.

toy, 2025-10-18:
@darkhz Did you test it? Anything unexpected about the behaviour?

darkhz, 2025-10-21:
@toy apologies, was very busy until yesterday. I can test it today or tomorrow, but the change you made will work regardless. I will let you know.

darkhz, 2025-10-21:
@toy update, I tested the branch on which the fix was applied (https://github.com/toy/blueutil/pull/new/fix-halt-on-unknown-disconnect), it works as expected. When I enter an invalid address now (with command `blueutil --disconnect <invalid-address>`), blueutil responds with a message "Device not connected '\<invalid-address\>'". Thank you for fixing this.

closed 2025-10-21 by darkhz

toy, 2026-09-09:
@darkhz Really sorry, I lost track of this issue and didn't merge and release it, so did it now as [v2.14.0](https://github.com/toy/blueutil/releases/tag/v2.14.0)